john2zy/IMU-Position-Tracking #2: Online using

zedxi, 2021-05-10:
Your project is really good.I'd like to use it for data processing in real-time. Is that work still well?Could you give me some advice for modifications. Thanks a lot

LibofRelax, 2021-05-12:
Yes you can make it work for real time processing. Just port the filter to the included real time code or remove filtering all together. Plotting in real time is not easy, maybe you can use Javascript stuff on a browser for that and send data with a http connection. However the tracking accuracy is not useable with 200 Hz sampling rate with my experiments. I'm not an expert but I'd say combing multiple IMUs would be helpful.

zedxi, 2021-06-13:
请问一下有实时运行的代码吗，因为发现卡尔曼滤波等函数需要用到全部数据，请问该如何修改呢?

LibofRelax, 2021-06-18:
不需要用到全部数据，while loop改为接收数据即可实时

closed 2021-09-02 by LibofRelax