john2zy/IMU-Position-Tracking #14: 6-DOF IMU which can output filtered acceleration without gravity and euler angles

xwhkkk, 2024-12-11:
Thank you for sharing your great job! I am working with an IMU sensor that outputs both gravity-filtered acceleration and Euler angles directly. I would like to inquire whether it is possible to adapt or modify your code to utilize this sensor output format. 
Could you kindly provide guidance or suggest which parts of the code would require changes?

Thank you for your time and support. I look forward to your response!

john2zy, 2024-12-11:
Sorry, that is way beyond me now. Acceleration is straight forward, you just skip the step that uses magnetic fields to subtract gravity out. For using Euler angles directly, I think the paper I included have formulas you can use.

xwhkkk, 2024-12-12:
Thank you for your kind reply.  I would like to know in which scenarios this code is best suited. Additionally, would it be feasible to use this for tracking and localizing wrist motion?